Seeed-Studio/Grove_LED_Bar #10: Raspberry Pi with wiringPi library

alexpagri, 2015-04-10:
I recently downloaded the library and edited it to run with wiringPi library.

I can only power up all leds but not toggle them or use any of the examples from this repo because the led bar won't change at all...

I have some questions :  Is there any delay between those "digitalWrite(x,y)" or "digitalRead(x)" ?
                                        Why when I swap the Clock and the Data pin and run an example the led changes but in a different, incorrect way ?

I have the LED Bar v1.0 version.

I can upload some code if it is needed...

I tried different ways, such as totally randoming the time for outputs, and it was kind of chaotic.....

I only need some explanation on how to use the Clock & Data pins to simply toggle a led and I mean in which order should the combination be ....

Thank you !

awong1900, 2015-05-19:
Hi @alexpagri ，

Are you solved it ?

It is ok on Arduino?

Ten

alexpagri, 2015-05-19:
@awong1900 

The problem is partially solved : 

I use a Raspberry pi 2 rev B and     in order to reset the leds ( change them ) , I have to cut the power. 

I cannot consider this a problem , I was just trying to set up the library successfully on my device.

Sorry for not updating the issue ....

I don't have an Arduino.

This issue can be closed .

closed 2015-05-20 by awong1900