Seeed-Studio/Grove_LED_Bar #24: ESP32

JonFehl, 2021-02-11:
Problems with esp32

JabberwockPL, 2021-06-13:
Yes, the examples do not compile for ESP32. Error log attached:

[Error2.txt](https://github.com/Seeed-Studio/Grove_LED_Bar/files/6643886/Error2.txt)

michhunger, 2021-06-29:
Hi I had the same problem. For me the problem was, that in the Grove_LED_Bar.cpp the min and max functions use default int 0 and 10 as inputs. While they compare with level which is a float. Apparently this is usually automatically casted, but somehow this didnt work. So you have to manually change the code so that everything is casted to a float then it should work. Later in the code they compare with an unsigned char there you have to cast to unsigned char. Hope this helps

JabberwockPL, 2021-06-29:
Thanks, I will check it out!

JabberwockPL, 2021-06-29:
This is the same issue and its solution:
https://github.com/AaronLiddiment/LEDSprites/issues/1

JabberwockPL, 2021-07-11:
Unfortunately, the project compiles, but the module still does not work with ESP32. It runs successfully three or four times and then just stops.

PR #25 (2021-09-15): Fix std::min() and std::max usage
This PR fixes issues with `std::min()` and `std::max()` usages:

```
Grove_LED_Bar.cpp:97:31: error: no matching function for call to 'min(int, float&)'
   97 |   level = max(0, min(10, level));
      |                               ^
Grove_LED_Bar.cpp:118:40: note:   mismatched types 'std::initializer_list<_Tp>' and 'float'
  118 |   brightness = max(0, min(brightness, 1));
      |                                        ^
```

Fixes #24
commit 60e426ee (2021-09-15): Fix std::min() and std::max usage

Fixes #24
closed 2021-09-15 by Pillar1989