Seeed-Studio/Grove_LED_Bar #29: not compatible with ESP

Luigi5555, 2022-03-24:
Several tests with Arduino UNO worked fine. But switching to "Board=Node-MCU/ESP-12E" resulted in numerous error messages, most of them in conjunction with "min" and "max". 

As I was told in another forum, there were also some warnings tgether with the "Mega" (not verified by myself). Please have a look also on that. 

Thanks in advance,
Luigi5555

[ErrorMessages.txt](https://github.com/Seeed-Studio/Grove_LED_Bar/files/8340270/ErrorMessages.txt)

JabberwockPL, 2022-03-25:
It was supposed to be fixed:

https://github.com/Seeed-Studio/Grove_LED_Bar/issues/24#issue-806260360

but it is still not working for me.

708yamaguchi, 2022-08-16:
@JabberwockPL 
This pull request may fix your problem.
https://github.com/Seeed-Studio/Grove_LED_Bar/pull/30

GuyIUT, 2024-09-23:
Currently, the library available on the Seedstudio website cannot be used for the ESP32 board with the Arduino IDE. When we try to compile all the examples, many errors appear on the min and max functions in Grove_LED_bar.cpp.
I have  solve all the problems by implementing parameter casting **(in bold)** in the function call. This file must be modified as follows:
line 97 level = max(0, min(10, **(int)** level));
line 117 led = max(1, min(10, **(int)** led));
line118 brightness = max(0, min(**(int)** brightness, 1));
line 138 led = max(1, min(10, **(int)** led));

ackPeng, 2024-10-16:
@JabberwockPL @Luigi5555 Do you still have any doubts about this issue?

JabberwockPL, 2024-10-16:
I did not have a chance to revisit this issue as my hardware requirements have changed in the meantime. For what it is worth, the example seems to compile now without problems when ESP32 is selected. I do not have the hardware to check whether it actually works.

ackPeng, 2024-10-16:
ok, I will close this issue. If you have any follow-up questions, please feel free to contact us.

closed 2024-10-16 by ackPeng