Seeed-Studio/Grove_LED_Bar #32: Library does not compile in Arduino IDE

Deja-Github, 2023-05-04:
I'm getting a compilation error when including this library in the Arduino IDE. I'm pretty new to Arduino and I'm not familiar with CPP, any ideas?

`In file included from C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.h:40:0,
                 from C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:33:
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp: In member function 'void Grove_LED_Bar::setLevel(float)':
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:40: error: expected ')' before numeric constant
   level = max(0.0F, min((MY9221_LED_NUM.0F), level));
                                        ^
C:\Users\rd\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:93:24: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                        ^
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:21: note: in expansion of macro 'min'
   level = max(0.0F, min((MY9221_LED_NUM.0F), level));
                     ^~~
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:53: error: expected ')' before ';' token
   level = max(0.0F, min((MY9221_LED_NUM.0F), level));
                                                     ^
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:53: error: expected ')' before ';' token
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:53: error: expected ')' before ';' token
C:\Users\rd\Documents\Arduino\libraries\Grove_LED_Bar\Grove_LED_Bar.cpp:97:53: error: expected ')' before ';' token

exit status 1

Compilation error: exit status 1`

Deja-Github, 2023-05-05:
update: I have tried compiling v1.0.1 of the lib with both Arduino IDE v1.8.19 and v2.1.0 and it consistently fails, however I just tried v1.0.0 of the library and that compiles ok.

ackPeng, 2024-10-16:
@Deja-Github Thank you very much for your question. We have updated this library. The reason for this problem is incorrect data type conversion. This problem should not occur now.

closed 2024-10-30 by Lesords